sparkfun/SparkFun_ICM-20948_ArduinoLibrary #134: Unable to set full-scale mode

LukeDeWaal, 2023-12-28:
### Unable to set full-scale mode
Full-Scale registers reset themselves instantaneously. I thought it was a delay problem, as with #8, but no matter the delay, it does not work.

### Your workbench
* Beaglebone Black Rev C
* ICM20948 from Pimoroni
* Connected using I2C-1 on the Beaglebone 
* Powered from USB
* I used the C backbone of the library and made wrappers to be identical to the CPP wrappers provided, only in C

### Steps to reproduce
Code is shown below

### Expected behaviour
Initialize DMP

### Actual behaviour
Full-scale mode registers fail to be set
```
2023/12/28-16:42:53  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 1 / 10
2023/12/28-16:42:54  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 2 / 10
2023/12/28-16:42:55  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 3 / 10
2023/12/28-16:42:56  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 4 / 10
2023/12/28-16:42:57  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 5 / 10
2023/12/28-16:42:58  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 6 / 10
2023/12/28-16:42:59  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 7 / 10
2023/12/28-16:43:00  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 8 / 10
2023/12/28-16:43:01  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 9 / 10
2023/12/28-16:43:02  WARN  ICM_20948_wrappers.c            :0173:       Failed to set full-scale mode: 10 / 10
2023/12/28-16:43:03  ERROR ICM_20948_wrappers.c            :0174:       Failed to set full-scale mode
2023/12/28-16:43:03  ERROR pru_manager_thread.c              :0083:       ICM20948 DMP init failed
```

### I2C signals
[image]

[image]

### C Wrapper Code:
```C
int icm20948_startupDefault(ICM_20948_Device_t* icm_dev, uint8_t minimal){
    int ret = 0;
    ICM_20948_Status_e res = ICM_20948_check_id(icm_dev);
    BBBLOG(error, res != ICM_20948_Stat_Ok, return -1, "ICM20948 WHOAMI test failed\n")

    res = ICM_20948_sw_reset(icm_dev);
    BBBLOG(error, res != ICM_20948_Stat_Ok, return -1, "ICM20948 SW Reset failed\n");

    usleep(50000);

    res = ICM_20948_sleep(icm_dev, false);
    BBBLOG(error, res != ICM_20948_Stat_Ok, return -1, "ICM20948 failed to exit sleep mode\n");

    usleep(1000);

    res = ICM_20948_low_power(icm_dev, false);
    BBBLOG(error, res != ICM_20948_Stat_Ok, return -1, "ICM20948 failed to exit low-power mode\n");

    usleep(1000);

    ret = icm20948_enable_magnetometer(icm_dev, minimal);
    if(ret != 0){
        return ret;
    }

    if(minimal){
        return 0;
    }

    res = ICM_20948_set_sample_mode(icm_dev, (ICM_20948_Internal_Acc | ICM_20948_Internal_Gyr), ICM_20948_Sample_Mode_Continuous);
[... 322 lines cut ...]
### Calling Code
```C
    ICM_20948_Serif_t icm_serif;
    icm_serif.read = &icm_read;
    icm_serif.write = &icm_write;
    ICM_20948_Device_t icm_dev;
    ICM_20948_init_struct(&icm_dev);
    ICM_20948_link_serif(&icm_dev, &icm_serif);
    res = icm20948_startupDefault(&icm_dev, 1);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 minimal setup failed\n");
    res = icm20948_initialize_DMP(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 DMP init failed\n");
    res = icm20948_enable_DMP_sensor(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 DMP sensor init failed\n");
    res = icm20948_set_DMP_ODR(&icm_dev, 10); // 5 Hz ODR
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 DMP ODR setting failed\n");
    res = icm20948_enable_FIFO(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 enable FIFO failed\n");
    res = icm20948_enable_DMP(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 enable DMP failed\n");
    res = icm20948_reset_DMP(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 reset DMP failed\n");
    res = icm20948_reset_FIFO(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 reset FIFO failed\n");
```

PaulZC, 2024-01-02:
Hi Luke (@LukeDeWaal ),

I'm not sure how we can help you here. This looks like a hardware issue between the Beaglebone and the ICM20948. Not an issue with this library?

I'm happy to leave this open in case anyone else can help.

Best wishes,
Paul

PaulZC, 2024-01-02:
Perhaps check the I2C signal voltages. If you are using a Saleae analyzer, try enabling the analog inputs. Check the voltages and the rise/fall times. Perhaps what the (digital) analyzer is reporting is different to what the Beaglebone receives?

LukeDeWaal, 2024-01-03:
Hi Paul,

> I'm not sure how we can help you here. This looks like a hardware issue between the Beaglebone and the ICM20948. Not an issue with this library?

I understand it isn't so much an issue with this library, but I figured most people active within this repository have a better chance of helping me with this issue. From what I've seen, you seem to know this repo better than your back pocket 😄 

> Perhaps check the I2C signal voltages. If you are using a Saleae analyzer, try enabling the analog inputs. Check the voltages and the rise/fall times. Perhaps what the (digital) analyzer is reporting is different to what the Beaglebone receives?

I don't own an oscilloscope or analog signal analyzer sadly, but I did manage to progress with this issue; in particular, after shortening the I2C lines the sensor behaved as expected. I suppose there was too much noise on the I2C bus. I will close this issue as it was a hardware fault all along! (Gotta love microelectronics eh 🙄 )

closed 2024-01-03 by LukeDeWaal