sparkfun/SparkFun_ICM-20948_ArduinoLibrary #135: DMP Quaternions

Amirhosein80, 2024-01-01:
Hello, I’m planning to use quaternion data from the ICM DMP module for motion capture purposes, and I have a few questions about DMP:
1. What are the differences between the INV_ICM20948_SENSOR_ROTATION_VECTOR and INV_ICM20948_SENSOR_ORIENTATION modes?
2. How is the sensor calibration and what does the Accuracy value represent in the sensor data?
3. When I convert the quaternions to roll, pitch, and yaw angles, the pitch value is constrained between 90 and -90 degrees. What might be the cause of this issue?

I would appreciate any insights or documentation you could provide on these matters.

Thank you!

PaulZC, 2024-01-02:
Hi @Amirhosein80 ,

1) INV_ICM20948_SENSOR_GAME_ROTATION_VECTOR is 32-bit 6-axis quaternion. INV_ICM20948_SENSOR_ROTATION_VECTOR is 32-bit 9-axis quaternion + heading accuracy. Please see: https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/DMP.md#which-dmp-features-are-currently-supported

2) For Accel, Gyro and Compass Accuracy: The accuracy is expressed as 0~3. The lowest is 0 and 3 is the highest.

3) This is normal. A pitch of 90 degrees represents "straight up" or "straight down".

Best wishes,
Paul

closed 2024-01-02 by PaulZC

Amirhosein80, 2024-01-27:
Thanks
I checked [https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/DMP.md#which-dmp-features-are-currently-supported], but there was no explanation about the difference in features.