sparkfun/SparkFun_ICM-20948_ArduinoLibrary #136: Latched WOM Interrupt Not Resetting

LukeDeWaal, 2024-01-04:
### Subject of the issue
When using WOM interrupts, when using the latched interrupt (instead of the 50us pulsed) the signal refuses to reset to low when using an active-high configuration. 

### Your workbench
* Platform: Beaglebone Black Rev C; Debian GNU/Linux 11 Bullseye 
* Wiring: ICM20948 and BME688 on I2C2 bus, interrupt going from ICM20948 to signal analyzer and GPIO input on Beaglebone Black
* Power: Through USB
* Are there any additional details that may help us help you?
    * The ARM processor uploads the DMP firmware and initializes the ICM20948
    * The PRU regularly retrieves samples from the ICM20948 and resets interrupts at an interval of 200ms
    * When using the pulsed (not latched) interrupt behaviour, a pulse train is generated but stops accordingly after no more motion occurs.

### Steps to reproduce

Code running on the PRU:
```C
#define PRU_I2C_READREG(DEV, SADDR, REG, DST, SIG, FLAGS, RES) \
    RES = pru_i2c_driver_ReadReg((DEV), (SADDR), (REG), (DST), &(FLAGS));   \
    (SIG).status_bits.flags = (FLAGS);                                             \
    if(!(RES)){                                                                               \
        __halt();                                                                           \
    }                                                                                       \
    FLAGS = 0;

int icm20948_clear_intc(void){ // This is called more or less every 200ms by the PRU to clear any interrupts
    icm20948_set_bank(0u);

    uint8_t tmp = 0;
    i2c_flags_t flags = 0;
    int res = 0;
    PRU_I2C_READREG(I2C_CHAN, ICM20948_SADDR, 0x19, &tmp, ICM20948_SIGNALS, flags, res);
    DelayMicros(100);
    PRU_I2C_READREG(I2C_CHAN, ICM20948_SADDR, 0x19, &tmp, ICM20948_SIGNALS, flags, res);
    return 0;

}
```
On the ARM processor, a thread is checking every 25 ms for events on the GPIO line and will send out a distress signal if a WOM interrupt is detected. The ICM20948 is initialized through:
```C
    ICM_20948_Device_t icm_dev;
    ICM_20948_init_struct(&icm_dev);
    icm_dev._dmp_firmware_available = true;
    ICM_20948_link_serif(&icm_dev, &icm_serif);
    res = icm20948_startupDefault(&icm_dev, 0);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 minimal setup failed\n");
    ICM_20948_smplrt_t samplerate_cfg = {.a = 19, .g = 19};
    uint16_t DMP_rate = 1100 / (1 + samplerate_cfg.g);
    // Gyro ODR is computed as follows: 1.1 kHz/(1+GYRO_SMPLRT_DIV[7:0]).     19 = 55Hz. InvenSense Nucleo example uses 19 (0x13).
    // Acc  ODR is computed as follows: 1.125 kHz/(1+ACCEL_SMPLRT_DIV[11:0]). 19 = 56.25Hz. InvenSense Nucleo example uses 19 (0x13).
    //mySmplrt.g = 4; // 225Hz
    //mySmplrt.a = 4; // 225Hz
    //mySmplrt.g = 8; // 112Hz
    //mySmplrt.a = 8; // 112Hz
    res = icm20948_initialize_DMP(&icm_dev, &samplerate_cfg);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 DMP init failed\n");
    res = icm20948_enable_DMP_sensor(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 DMP sensor init failed\n");
    res = icm20948_set_DMP_ODR(&icm_dev, 5, DMP_rate); // 5 Hz ODR
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 DMP ODR setting failed\n");
    res = icm20948_enable_FIFO(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 enable FIFO failed\n");
    res = icm20948_enable_DMP(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 enable DMP failed\n");
    res = icm20948_reset_DMP(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 reset DMP failed\n");
    res = icm20948_reset_FIFO(&icm_dev);
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 reset FIFO failed\n");
    res = icm20948_cfg_int(&icm_dev, 0, 1, 1); // Active low = 0, open-drain = 1, latch = 1
    BBBLOG(error, res != 0, goto cleaning, "ICM20948 int cfg failed\n");
    res = icm20948_set_WOM(&icm_dev, 1, 1, 0x32); // Enable WOM, compare consecutive samples, and thresh @ 0x32 = 200 mg, 0xFF = 1020 mg, 0x01 = 4 mg
[... 421 lines cut ...]

    ICM_20948_INT_enable_t enreg;
    status  = ICM_20948_int_enable(icm_dev, NULL, &enreg);
    enreg.DMP_INT1_EN = false;
    enreg.WOM_INT_EN = true;
    status |= ICM_20948_int_enable(icm_dev, &enreg, &enreg);
    BBBLOG(error, status != ICM_20948_Stat_Ok, return -1, "Failed to enable WOM interrupt\n");

    return 0;
}

int icm20948_clear_interrupts(ICM_20948_Device_t* icm_dev){
    ICM_20948_INT_STATUS_t int_stat;

    // read to clear interrupts
    ICM_20948_Status_e status = ICM_20948_set_bank(icm_dev, 0);
    BBBLOG(error, (status != ICM_20948_Stat_Ok), return -1, "Failed to set register bank\n");

    status = ICM_20948_execute_r(icm_dev, AGB0_REG_INT_STATUS, (uint8_t *)&int_stat, sizeof(ICM_20948_INT_STATUS_t));
    BBBLOG(error, (status != ICM_20948_Stat_Ok), return -1, "Failed to read interrupt status register 0\n");

    return status;
}

```

LukeDeWaal, 2024-01-17:
@PaulZC  do you maybe have any idea as to why the interrupt is not resetting after clearly reading the INT_STATUS register and seeing that the INT_STATUS register says the interrupt is cleared?

PaulZC, 2024-01-18:
Hi Luke (@LukeDeWaal ),

In [clearInterrupts](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9a10c510ddb694f08aa93c12d586358cb45abd2b/src/ICM_20948.cpp#L716) we read both INT_STATUS registers. I see you are just reading one. Maybe that's it?

Also, please check you are using the correct register bank. Please make sure you are replicating [these lines](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9a10c510ddb694f08aa93c12d586358cb45abd2b/src/ICM_20948.cpp#L1753) from the Arduino ```begin```. Maybe something is pointing at the wrong register bank?

I hope this helps,
Paul

LukeDeWaal, 2024-01-18:
> In [clearInterrupts](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9a10c510ddb694f08aa93c12d586358cb45abd2b/src/ICM_20948.cpp#L716) we read both INT_STATUS registers. I see you are just reading one. Maybe that's it?

I tried this but sadly that's not the issue. The odd thing is that the bits in INT_STATUS are definitely cleared, but the actual interrupt line is not pulled back down for some reason. I also tried this by turning on the `INT_ANYRD_2CLEAR` bit but this also does not help. 
 
> Also, please check you are using the correct register bank. Please make sure you are replicating [these [...]

In my code I have a static variable inside the `set_bank` function that keeps track of the last memory bank:
```C
int icm20948_set_bank(uint8_t bank){
    static uint8_t last_bank = UINT8_MAX;
    if(last_bank == bank){
        return 0;
    }
    uint16_t flags = 0;
    int res;
    uint8_t out = ((bank & 0b11) << 4);
    PRU_I2C_WRITEREG(I2C_CHAN, ICM20948_SADDR, 0x7F, out, ICM20948_SIGNALS, flags, res);
    last_bank = bank;
    return 0;
}
```
which achieves the same result I believe. 

When I set the pin configuration I set it to active-high, push-pull, and latched (I set the GPIO port that poll's for the rising edge on my board as push-pull and active-high as well).

PaulZC, 2024-01-29:
Closing... Please re-open if you need more help with this.

closed 2024-01-29 by PaulZC