sparkfun/SparkFun_ICM-20948_ArduinoLibrary #137: Storing and setting Compass Calibraiton

nyfelix, 2024-01-23:
Dear Team,

first of all thank you for the development efforts. This seems to be the only available library that manages to handle the DMU. As I understand the ICM-20949 does selfcalibration and using the mple6_DMP_Quat9_Orientation.ino sketch I can observe that the orientation stabalizes after few seconds of moving the sensor.

No I want to store the calibration and set it at starup of my firmware (so not moving of the sensor is required).
From what I understood the following code will output the current calibraiton:

`
icm_20948_DMP_data_t data;
myICM.readDMPdataFromFIFO(&data);
Serial.print(data.Compass_Calibr.Data.X);
`

Is there any method that allows me to set this calibraiton at startup?

Best,
Felix

PaulZC, 2024-01-23:
Hi Felix (@nyfelix ),

Please see [this example](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example11_DMP_Bias_Save_Restore_ESP32/Example11_DMP_Bias_Save_Restore_ESP32.ino). It is ESP32-specific; the biases are stored in ESP32 EEPROM. Please adapt the storage to match your platform.

Best wishes,
Paul

closed 2024-01-23 by PaulZC

nyfelix, 2024-01-23:
Perfect! That worked :-) 
Sorry I did not see the example myself.

PaulZC, 2024-01-23:
No worries Felix - glad that's working for you.
Best,
Paul