sparkfun/SparkFun_ICM-20948_ArduinoLibrary #141: Magnetometer Calibration

jasvinder1079, 2024-03-06:
Hi,
I have been using example 7 i.e. Quat-6_Eulersangles(with dmp enabled) and getting correct values of roll and pitch but there is some issues with yaw values (slightly unstable with roll and pitch yaw also changing).That's why I selected eg 6 Quat-9 with 
INV_ICM20948_SENSOR_ORIENTATION for better yaw values but in this example with the little change in roll value the yaw values changing from 40 to 160 degree suddenly .Is this because of not calibration ?Do we need to some calibration before using this or dmp doing it automatically? 
Im using icm-20948 with RPi 4.

PaulZC, 2024-03-06:
Hi Jasvinder (@jasvinder1079 ) ,

Please search the open and closed issues. Many of them are relevant to DMP (IMU) calibration.

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues?q=is%3Aissue+is%3Aclosed+calibrate+rotate

Example 7 uses the Game Rotation Vector (6-axis, no magnetometer). Example 6 uses Sensor Orientation (9-axis, with magnetometer). I think Sensor Orientation is probably the best choice for your application.

Best wishes,
Paul

Best wishes,
Paul

closed 2024-03-06 by PaulZC

jasvinder1079, 2024-03-08:
thanks @PaulZC 
I'm using this imu for surveying purpose (tilt compesation), As suggested i rotated the device in every direction for 10-20 second and hold it for few seconds in each axes ,But my yaw giving wrong .
I just want to know how we get to know its calibrated (for how much time we have to do this calibration and how often), as in land survey we need to use device for 3-5 hrs continuously ,do we only need to calibrate it once while powering on ?

PaulZC, 2024-03-08:
Hi Jasvinder (@jasvinder1079 ),

I am sorry, I do not know if the 20948 DMP is suitable for this. You will need to test it yourself. The sensors will drift over time and with temperature.

There are IMU chips designed for tilt compensation. Please search "FMI IM19".

I hope this helps,
Paul