sparkfun/SparkFun_ICM-20948_ArduinoLibrary #143: Drift in ICM20948 IMU.

abhilashv-ign, 2024-06-05:
### Subject of the issue
Describe your issue here.

### Your workbench
* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

abhilashv-ign, 2024-06-05:
I'm using the ICM20948 IMU for getting the angular rotation of a robot. I'm using complementary filter for calculating the q values and the viasualization is done using rviz2. 
The accurracy of the IMU sensor seems to be on point. When i rotate the IMU 90 degrees I'm getting 90degree rotation in visualization as well. The deviation in performance is when I'm keeping the IMU at rest.
The visualization is supposed be stable and not moving. But the output that I'm getting is a slight drift. i.e the visualization rotates about z-axis over time. The rate of rotation is minute but over the course of a minute a significant drift is visible.
I have tried adjusting the sensitivity as well. 
The DMP is disabled and the sampling mode is set to continuous mode.

Thank you for the suggestions.

nemccarthy, 2024-06-18:
Did you have any luck with this issue?

timonkobusch, 2024-06-26:
Maybe I understand something incorrectly, but the gyrosensor will always have a drift see: [https://electronics.stackexchange.com/questions/445787/what-actually-causes-gyroscope-drift-in-imu](https://electronics.stackexchange.com/questions/445787/what-actually-causes-gyroscope-drift-in-imu)
I am not sure about calibrating the hardware, but for the project of my bachelor thesis, we programmed a kalman Filter (also sperate Particle Filter) in combination with the acceleration sensor to cancel out the drift. I think it is the common approach to handle the gyrosensor drift.