sparkfun/SparkFun_ICM-20948_ArduinoLibrary #146: Example 11 double initiation of ICM_20948::setFullScale

Gord1, 2024-08-10:
Using Example 11

I'm trying to figure out what the scale is set for the gyro so I can scale the output properly.

It appears during setup() the ICM_20948::setFullScale is called twice through two different dmp initialization routines and the full scale values for acceleration and gyro are different in each section.  I don't think that is intended.

In one case :
ICM_20948::setFullScale(uint8_t, ICM_20948_fss_t) : ICM_20948_Status_e
	ICM_20948::initializeDMP() : ICM_20948_Status_e
		setup() : void
with 
  myFSS.a = gpm4;        // (ICM_20948_ACCEL_CONFIG_FS_SEL_e)
                         // gpm2
                         // gpm4
                         // gpm8
                         // gpm16
  myFSS.g = dps2000;     // (ICM_20948_GYRO_CONFIG_1_FS_SEL_e)
                         // dps250
                         // dps500
                         // dps1000
                         // dps2000

but it is also called by:
ICM_20948::setFullScale(uint8_t, ICM_20948_fss_t) : ICM_20948_Status_e
      ICM_20948::startupDefault(bool) : ICM_20948_Status_e
		ICM_20948_I2C::begin(TwoWire &, bool, uint8_t) : ICM_20948_Status_e
			setup() : void
with
  ICM_20948_fss_t FSS;
  FSS.a = gpm2;   // (ICM_20948_ACCEL_CONFIG_FS_SEL_e)
  FSS.g = dps250; // (ICM_20948_GYRO_CONFIG_1_FS_SEL_e)

I'm not sure why an I2C begin statement should set accel and gyro FS values ...

The FSS is buried deep in the code and is difficult to find.

PaulZC, 2024-08-12:
Hi @Gord1 ,

If you are using the DMP, you need to set the Gyro FSR to 2000dps and the Accel FSR to 4g. This is defined in the confidential InvenSense DMP Application Note - which we cannot share with you.

If you are not using the DMP, you can set the FSS to whatever range works best for your application. We use dps250 and gpm2 in non-DMP examples.

Best wishes,
Paul

closed 2024-08-12 by PaulZC

Gord1, 2024-08-12:
Hi @PaulZC 
Thanks for this information. 

Could you add a note to the code that this is a requirement for DMP so the next person doesn't fall into the same trap?

Thanks for your work.

Gord

vlacorre, 2024-08-22:
Hi @PaulZC,

Thank you for the clarification.
However you said it was possible to modify the Accel FSS when using the DMP in this issue: https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/107#issuecomment-1322073510.
By following this, I managed to change FSS on the Accel, but not the Gyro. Could you confirm that it is not possible?

Best wishes,
Vivien

PaulZC, 2024-08-22:
Hi Vivien (@vlacorre ),

Thank you - I had forgotten about that example!

For the Gyros, there is the scaling factor GYRO_SF - which needs to be calculated based on the sample rate. Please see [inv_icm20948_set_gyro_sf](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/168c18b8ebf7c8a28ac22f1351f23e0f143ba195/src/util/ICM_20948_C.c#L2572-L2630) which is called by [setGyroSF](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/168c18b8ebf7c8a28ac22f1351f23e0f143ba195/src/ICM_20948.cpp#L1439-L1452). The InvenSense application note doesn't mention GYRO_FULLSCALE. We got that code from their code examples...

Maybe ```setGyroSF``` should be called _after_ setting GYRO_FULLSCALE? Looking at the [captured DMP configuration](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/DMP.md#can-i-see-the-full-dmp-configuration-captured-from-zanel-s-teensy-icm-20948-code), GYRO_FULLSCALE and GYRO_SF are set multiple times. Maybe this is to allow TIMEBASE_CORRECTION_PLL to update, before the GYRO_SF is set for the final time?

There are many things about the DMP we still do not understand...

Best wishes,
Paul