sparkfun/SparkFun_ICM-20948_ArduinoLibrary #150: Question about DMP activation code  inv_icm20948_enable_dmp_sensor

dobodu, 2024-11-09:
I believe the code activating too much Data_ready an Motion Event (but the more can do the leas).

I explain :

`delta = 0; // Clear delta
  for (int i = 0; i < 32; i++)
  {
    androidSensorAsBitMask = 1L << i;
    if ((pdev->_enabled_Android_0 & androidSensorAsBitMask) > 0) // Check if the Android sensor (0-31) is enabled
    {
      delta |= inv_androidSensor_to_control_bits[i]; // If it is, or the required bits into delta
    }
    if ((pdev->_enabled_Android_1 & androidSensorAsBitMask) > 0) // Check if the Android sensor (32-) is enabled
    {
      delta |= inv_androidSensor_to_control_bits[i + 32]; // If it is, or the required bits into delta
    }
    // Also check which bits need to be set in the Data Ready Status and Motion Event Control registers
    // Compare to INV_NEEDS_ACCEL_MASK, INV_NEEDS_GYRO_MASK and INV_NEEDS_COMPASS_MASK
    if (((androidSensorAsBitMask & INV_NEEDS_ACCEL_MASK) > 0) || ((androidSensorAsBitMask & INV_NEEDS_ACCEL_MASK1) > 0))
    {
      data_rdy_status |= DMP_Data_ready_Accel;
      inv_event_control |= DMP_Motion_Event_Control_Accel_Calibr;
    }`

for i = 0 to 32 will check all the time every bits for INV_NEEDS_ACCEL_MASK (not null) so data will always be activated. 

I believe the correct test should be 

if (((pdev->_enabled_Android_0 & androidSensorAsBitMask & INV_NEEDS_ACCEL_MASK) > 0) || ((pdev->_enabled_Android_1 & androidSensorAsBitMask & INV_NEEDS_ACCEL_MASK1) > 0))

PaulZC, 2024-11-11:
Hi @dobodu ,

Nice work - thank you! Yes, I think you are correct:

Looking again at the ICM20948_eMD_nucleo_1.0 example code, I can see that InvenSense used a method ```inv_convert_androidSensor_to_control``` to set the two ```s->inv_androidSensorsOn_mask``` masks. Only if the mask matches ```INV_NEEDS_GYRO_MASK``` etc. is the ```data_rdy_status``` bit set.

I will change this in the release_candidate branch and will test it.

Best wishes,
Paul

PaulZC, 2024-11-11:
Thank you @dobodu - this is resolved in v1.3.0

closed 2024-11-11 by PaulZC

dobodu, 2024-11-11:
You are welcome. I'm porting the library to Micropython so I dig a bit in the code...

Another thing I wonder :

When setting up Low Power Mode with setSampleMode (in DMP activation), it uses AGB0_REG_LP_CONFIG (in set sample mode),
the technical datasheet says :

Bit 0 to 3 are reserved
Bit 4 is for Gyro_cycle
Bit 5 is for Acccel Cycle
Bit 6 is for I2C_Mst_cycle

However ICM_20948_C.h has a typedef enum of
ICM_20948_Internal_Acc = (1 << 0),
ICM_20948_Internal_Gyr = (1 << 1),
ICM_20948_Internal_Mag = (1 << 2),
ICM_20948_Internal_Tmp = (1 << 3),
 ICM_20948_Internal_Mst = (1 << 4) }
 
so when turning LP  I2C_Master (1<<4 = 0b10000), the 5th bit is elevated but is corresponding to Acccel Cycle

I am true ?

PaulZC, 2024-11-12:
Hi @dobodu ,

The ```ICM_20948_InternalSensorID_bm``` bit definitions (```ICM_20948_Internal_Acc```, ```ICM_20948_Internal_Gyr```) are not written to AGB0_REG_LP_CONFIG directly. Instead, they are used to set the correct bits in ```ICM_20948_LP_CONFIG_t``` and then that is written to AGB0_REG_LP_CONFIG:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9caa80c79dd510039ec1680f8ac88a2b0f0289c2/src/util/ICM_20948_C.c#L647-L658

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9caa80c79dd510039ec1680f8ac88a2b0f0289c2/src/util/ICM_20948_REGISTERS.h#L225-L232

Best,
Paul

dobodu, 2024-11-12:
Thank you Paul,
I should take advice to read twice before posting, especialy after midnight.
Cheers !

dobodu, 2024-11-13:
Hi Marc,

I realeased a first version of ICM20948 Micropython I2C Library with DMP support
[https://github.com/dobodu/ICM20948_DMP_Micropython](https://github.com/dobodu/ICM20948_DMP_Micropython)