sparkfun/SparkFun_ICM-20948_ArduinoLibrary #152: Question about DMP Mount Matrix

dobodu, 2024-11-16:
Hi mark,

Once again, thank you for this piece of work, I was soo much inspirated by your code for my Micropython Library.

I am wondering on the way you set the mount matrix in your library. Let me exmplain.

The mounting matrix is a rotation matrix where normally Q.tQ = 1 (in other words determinant = 1)
I totally understand the way your library works, adding a scale factor to output directly in uT (versus hardware unit).

However, if the rotation matrix can be expressed as 9x9 32bits numbers, the DMP compass output is a 3-tupple of 16 bits.
For example, X-axis (M00 factor) is 0x09999999 so whenever the raw magnetometer data is 0x01, 0x01 * 0x09999999 lead
to an output > DMP compass register capacity.

I am wrong (I might missed something, and I'm a bit lazy do dig in Inversense Nucleo code)

PaulZC, 2024-11-18:
Please see [DMP.md](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/DMP.md#can-i-see-the-full-dmp-configuration-captured-from-zanel-s-teensy-icm-20948-code)

[image]

For more details, you will need to dig into the InvenSense Nucleo code...

PaulZC, 2024-11-18:
```
/* 0.15 uT * (1 << 30) */
#define DATA_AK09916_SCALE       161061273
```

161061273 = 0x09999999

dobodu, 2024-11-20:
Thank you Mark,
I've been digging a bit more and understand I shouldn't mix DMP "inside mem register" and DMP "outputs.

I've made many correction to le Micropython Library, but I'm faing an issue where the magnetometer outputs throught
DMP are incorrect (accelerometer and gyro are working well).

I got a strange Mx -1 My -1 Mz 514 and when I move the device, values are increasing of decreasing by 256 (2^8)....
I've chacked and checked again the code but I dont find the issue. Any ideas ?

PaulZC, 2024-11-21:
Hi @dobodu ,

My name is Paul - not Mark.

Check you are replicating [this code](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9caa80c79dd510039ec1680f8ac88a2b0f0289c2/src/ICM_20948.cpp#L1475-L1521) and [this code](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9caa80c79dd510039ec1680f8ac88a2b0f0289c2/src/ICM_20948.cpp#L1533-L1535) correctly. The magnetometer is an I2C peripheral. If you don't configure it correctly, you get very strange results. It took me days to figure all of this out. To make sure your code is working correctly, you may have to capture the bus traffic and go through it byte by byte, like I did in DMP.md.

Best,
Paul

dobodu, 2024-11-21:
Hi Paul (so sorry yesterday too much code reading).
Yes I figured out to follow step by step (by the way that was what I did to rewrite mycropython lib).
For shure I missed something 
 I think i'll use my debugging lines to output every action and follow line by line...

dobodu, 2024-11-24:
Well, I digged but I couldn't find what's going wrong.

Can you please tell me what report do you get when requiring DMP to report "INV_ICM20948_SENSOR_MAGNETIC_FIELD_UNCALIBRATED" ?

PaulZC, 2024-11-25:
If I run [Example9](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino), rotate the sensor around all three axes, orient the sensor North and Down (to try and align Y to maximum magnetic field), I see:

```
08:36:03.288 -> Q1:-0.141 Q2:-0.358 Q3:0.920
08:36:03.475 -> Q1:-0.141 Q2:-0.358 Q3:0.920
08:36:03.663 -> Q1:-0.141 Q2:-0.357 Q3:0.920
08:36:03.663 -> Accel: X:-1056.00 Y:-5056.00 Z:6532.00
08:36:03.663 -> Gyro: X:7.00 Y:45.00 Z:6.00
08:36:03.663 -> Compass: X:-25.00 Y:-182.00 Z:22.00
```

But if I change the orientation so that Z aligns with the field, I see:

```
08:52:04.150 -> Q1:-0.146 Q2:-0.901 Q3:0.409
08:52:04.337 -> Q1:-0.146 Q2:-0.901 Q3:0.408
08:52:04.525 -> Q1:-0.146 Q2:-0.901 Q3:0.408
08:52:04.525 -> Accel: X:-891.00 Y:-6148.00 Z:-5524.00
08:52:04.525 -> Gyro: X:-25.00 Y:13.00 Z:0.00
08:52:04.525 -> Compass: X:-35.00 Y:-3.00 Z:-378.00
```

The field value here should be approximately 50uT.

From my notes: ```// The AK09916 produces a 16-bit signed output in the range +/-32752 corresponding to +/-4912uT. 1uT = 6.66 ADU.```

If the compass values are 'raw', (50 / 4912) * 32752 would be approximately 333.

I hope this helps.

dobodu, 2024-11-25:
> If I run [Example9](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino), rotate the sensor around [...]

Thank you.

I think I'll dig with the way I write to DMP memory. I think the way I write DMP memory bank (only write when change) and register might be incorrect (I think I've seen that in programming reference) this might lead to partial writes and maybe explain my issue... 

Exemple 

    #Switch memory bank
    def DMP_bank(self, value):
        if not self._membank == value:
            self.write(0, ICM_MEM_BANK_SEL, value)
            self._membank = value

dobodu, 2024-11-30:
Hi Paul.

2 weeks that I try to find what's wrong with my library. I've been following line by line the original C/C++ one, but I'm shure I miss something. I've been rewriting some procedure to make them looking like yours.
I'm pretty shure the code is right (I mean no write error or kind of stuff) since FIFO outputs Accelerometer and Gyro.
But I can't find what's wrong with

I'll appreciate if you could give just a look to the DMP Initialisation I did  ( def DMP_config(self) ). For shure there must be an elephant in the corridor ;-)

dobodu, 2024-12-11:
Hi Paul, instead of spamming you repo, I detailled precisely my code in the issue I got, it's easy to read even if not python friendly.
I would appreciate you give me your opinion.
Thanks a lot

dobodu, 2024-12-15:
I FOUND IT !

The error was in no way related to DMP or compass mount matrix.

Let(s explain my mystage (for Pyhon User)

`from struct import unpack_from

DMP_Pedom_Quat6_Byte_Ordering = [1, 0, 3, 2, 5, 4]
DMP_Compass_Bytes = 6

print()
print("First Method Error")
print("==================")

data = bytearray(b'\xff\x92\xff\xf1\x02\x7f')
data_ordered = data
print("Raw data", data)
for i in range(DMP_Compass_Bytes) :
    data_ordered[DMP_Pedom_Quat6_Byte_Ordering[i]] = data[i]
print("Ordered data", data_ordered)
mx,my,mz = unpack_from("<3h", data_ordered)
print(mx, my, mz)

print()
print("Second Method Success")
print("=====================")

data = bytearray(b'\xff\x92\xff\xf1\x02\x7f')
data_ordered = bytearray(DMP_Compass_Bytes)
print("Raw data", data)
for i in range(DMP_Compass_Bytes) :
    data_ordered[DMP_Pedom_Quat6_Byte_Ordering[i]] = data[i]
print("Ordered data", data_ordered)
mx,my,mz = unpack_from("<3h", data_ordered)
print(mx, my, mz)`

First Method Error
==================
Raw data bytearray(b'\xff\x92\xff\xf1\x02\x7f')
Ordered data bytearray(b'\xff\xff\xff\xff\x02\x02')
-1 -1 514

Second Method 
==================
Raw data bytearray(b'\xff\x92\xff\xf1\x02\x7f')
Ordered data bytearray(b'\x92\xff\xf1\xff\x7f\x02')
-110 -15 639

dobodu, 2024-12-15:
However, Thank you Paul, for help !

PaulZC, 2024-12-16:
Hi @dobodu ,

Apologies for the slow reply - I was away on vacation.

Thank you for the update - I am glad the DMP is working for you. Copying in Python has tripped me up a few times - usually when copying dictionaries or lists. ```import copy``` and ```copy.deepcopy()``` can avoid a lot of pain...!

Best,
Paul