sparkfun/SparkFun_ICM-20948_ArduinoLibrary #154: Example 11 Issue

shivenpatel399, 2025-03-22:
Hey!

I tried running Example 11 (https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example11_DMP_Bias_Save_Restore_ESP32/Example11_DMP_Bias_Save_Restore_ESP32.ino) after doing Example 1 and 2, and I realize there was an issue regarding EEPROM.h where it is not in the library. I tried looking all over the internet for this library, but could not find the right one.

PaulZC, 2025-03-24:
Hi Shiven (@shivenpatel399 ),

Welcome!

Many Arduino boards support "EEPROM" storage. But it varies from board to board. I think I was using arduino-esp32 when I wrote Example11. EEPROM.h is part of the ESP32 board package, not this library.

In the Arduino IDE, have a look at the File \ Examples \ "Examples for <your board>". If it supports EEPROM, there should be an example in there which you can use and adapt. Note that some EEPROM libraries need a ".begin()" before storage can be accessed. Have a good look at the board example, and copy the relevant lines into Example11.

Good luck!

I'm going to close this issue. But please ask more questions here if you need more help.

Best wishes,
Paul

closed 2025-03-24 by PaulZC

shivenpatel399, 2025-03-24:
Thanks Paul!