sparkfun/SparkFun_ICM-20948_ArduinoLibrary #155: Error LoadProhibited when package init function

nijng21, 2025-04-03:
**Subject of the issue**
Hi @PaulZC , very thank you for your code about the ICM20948 on the Arduino and I want to discuss issue which i meet.

PaulZC, 2025-04-03:
Hi @nijng21 ,

You are mixing the use of `Serial` and `SERIAL_PORT`. Is the "Guru Meditation Error" caused by printing to an uninitialized port?

```
Serial.println("Initializing ICM20948...");
```

I hope this helps,
Paul

PaulZC, 2025-04-03:
Try using the [ESP32 Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder). It will tell you which line of code is causing the error.

I am closing this because I am confident the issue is in your code, not this library.

Best wishes,
Paul

closed 2025-04-03 by PaulZC

nijng21, 2025-04-03:
Hi @PaulZC ,

I have done it now.
I copy the error code in the Exception Decoder and i want to share  the result  with you.

[image]

Best,
Renhao

PaulZC, 2025-04-03:
Hi Nyplace (@nijng21 ),

OK. So it is failing at line 151, during the write:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9caa80c79dd510039ec1680f8ac88a2b0f0289c2/src/util/ICM_20948_C.c#L151

This probably means that the SPI port is not initialized correctly. You probably have an error in `SPI_PORT`? Please check ```myimu.init_spi(CS_PIN, &SPI_PORT);```. Should it be ```myimu.init_spi(CS_PIN, SPI_PORT);``` ?

Best,
Paul

PaulZC, 2025-04-03:
Please be careful with your SPI addresses and pointers.

Should

```void ICM::init_spi(uint8_t csPin, SPIClass *spiPort, uint32_t SPIFreq)```

be

```void ICM::init_spi(uint8_t csPin, SPIClass &spiPort, uint32_t SPIFreq)```

?

nijng21, 2025-04-06:
Hi @PaulZC ,

I'm sorry for i can't reply in time.
I have question about the definition of init function whether SPIClass must be define to &. When i define it to *, and below begin function modify to begin(csPin, *spiPort). I think may it be like effect.
Unfortunately, at first i define function like
`void ICM::init_spi(uint8_t csPin, SPIClass &spiPort, uint32_t SPIFreq)`
it is same error in the core.

Best,
Nyplace